Networking-for-Arduino/EthernetENC #1: Ping

opensprinklershop, 2020-09-29:
Howto use ping with EthernetENC? 
I mean send icmp requests

jandrassy, 2020-10-06:
sorry for the delay.
uIP supports ICMP, but it is not implemented in Ethernet. it should be possible. 
I will try to implement it later this week like this
https://www.arduino.cc/en/Reference/WiFiNINAPing

jandrassy, 2020-10-11:
so uIP only supports replaying to ping, it doesn't support recieve of replay to ping. (sending ping is simple)
I did find icmp enhancement of uIP only for IPv6. it was written by Adam Dunkels the author of uIP. it started as implementation for icmp6 replay and then it was enhanced for sending a ping request. Maybe I could add IPv4 ICMP the same way.

opensprinklershop, 2020-10-16:
Maybe this implementation could help? It's for the W5100:
https://github.com/BlakeFoster/Arduino-Ping

jandrassy, 2020-10-17:
> Maybe this implementation could help? It's for the W5100:
> https://github.com/BlakeFoster/Arduino-Ping

no. on W5c00 the TCP/IP stack is in the chip. here I have to enhance ICMP support to the 20 years old uIP library, which is merged in this library

opensprinklershop, 2020-12-28:
Or this?
https://github.com/ValentinBELYN/icmplib/tree/master/icmplib

closed 2020-12-28 by jandrassy