Networking-for-Arduino/EthernetENC #3: problems with IP in sending UDP packets

Andyroc, 2020-10-22:
Spent several hours and still can't find how to do it. I can not send UDP to a custom IP address, only to the one with which a connection is established. Maybe there is some kind of filter that does not allow this, or somewhere I can change the destination IP manually in udp packet?

jandrassy, 2020-10-23:
try the github version.

Andyroc, 2020-10-28:
I solved my problem, by disabling this part in uip_arp.c

/*
    if(i == UIP_ARPTAB_SIZE) {
     // The destination address was not in our ARP table, so we
	//// overwrite the IP packet with an ARP request. 

      memset(BUF->ethhdr.dest.addr, 0xff, 6);
      memset(BUF->dhwaddr.addr, 0x00, 6);
      memcpy(BUF->ethhdr.src.addr, uip_ethaddr.addr, 6);
      memcpy(BUF->shwaddr.addr, uip_ethaddr.addr, 6);
    
      uip_ipaddr_copy(BUF->dipaddr, ipaddr);
      uip_ipaddr_copy(BUF->sipaddr, uip_hostaddr);
      BUF->opcode = HTONS(ARP_REQUEST); //// ARP request. 
      BUF->hwtype = HTONS(ARP_HWTYPE_ETH);
      BUF->protocol = HTONS(UIP_ETHTYPE_IP);
      BUF->hwlen = 6;
      BUF->protolen = 4;
      BUF->ethhdr.type = HTONS(UIP_ETHTYPE_ARP);

      uip_appdata = &uip_buf[UIP_TCPIP_HLEN + UIP_LLH_LEN];
    
      uip_len = sizeof(struct arp_hdr);
      return;
    }

  */

jandrassy, 2020-10-28:
there is a less invasive way for this patch, I think.
instead of removing code in uip_arp.c remove the line uip_arp_out() from UIPUDP::_send in EthernertUdp.cpp

I already looked at this and I have to investigate it to understand why the switch from normal packet to arp packet is dome differently for UDP then TCP