Networking-for-Arduino/EthernetENC #4: Not compiling: 'class EthernetServer' has no member named 'write'

byteptr, 2020-12-07:
I take this source [https://www.arduino.cc/en/Tutorial/LibraryExamples/ChatServer]
Then following these notes seems not compile: 
[https://github.com/jandrassy/EthernetENC/wiki/Examples](url)

"Only change #include <Ethernet.h> to #include <EthernetENC.h>." <-- I did, and this is OK
"In ChatServer and DhcpChatServer examples please use EthernetServerWrite server(23)" <-- I replaced the line "EthernetServer server(23);" with "EthernetServerWrite server(23);" but I have the following error: 'EthernetServerWrite' does not name a type; did you mean 'EthernetServerPrint'?

Can you provide a modified example compiling please ?

jandrassy, 2020-12-08:
https://github.com/jandrassy/EthernetENC/wiki/Examples

jandrassy, 2020-12-08:
fixed in Wiki

closed 2020-12-08 by jandrassy