Networking-for-Arduino/EthernetENC #6: Ethernet.begin(mac) not working in ESP32

arangu97, 2021-02-18:
Hi, 

I am having some troubles in getting Esp32 working with ENC28J60 module. 
The Ethernet.begin function gets stucked, but I am not sure that the Ethernet.init function is working properly before of that.

Is there any way to check if the module has been initiated properly? I am using the GPIO_5 pin of ESP32 for the CS pin:

Ethernet.init(GPIO_NUM_5)

However if I check the Ethernet.hardwareStatus() is returning EthernetNoHardware.

I think that maybe the ENC28J60 is not even initalized.

Appreciate response.

jandrassy, 2021-02-18:
what example do you try?

arangu97, 2021-02-18:
Hello,

I have tried the DHCPAddressPrinter example. 

I have wire the ESP32 with the module like in this diagram : 

[image]

I have checked that those are the default SPI pins for the ESP32.

Thanks for your response

jandrassy, 2021-02-18:
try to disable WiFi or if the ENC module can be powered with 5V power it with 5 V.
and try to wire reset pin too.

arangu97, 2021-02-18:
I have also tried to supply it with 5V and with the reset pin wired. But is not working.

Is there any way to reset the module programatically? 

I attach the code, in case you see any error or something

[ethernet_test.zip](https://github.com/jandrassy/EthernetENC/files/6004163/ethernet_test.zip)

jandrassy, 2021-02-18:
try byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

arangu97, 2021-02-22:
Nothing, it keeps getting stucked in Ethernet.begin(mac).

Why would a change in the MAC solve my problem?

Thanks, but the Ethernet example is still not working

jandrassy, 2021-02-22:
could you try the WebServer example with a correct static IP?

arangu97, 2021-02-23:
Sorry, but nothing either: 

The following messages appear in the Serial port : 

_Ethernet WebServer Example
Ethernet shield was not found.  Sorry, can't run without hardware. :(_

I have connected to my gateway in order to see if it detects the ESP32 through the ENCJ module but it does not appear.

I have checked also to connect with other devices to the router and the DHCP is activated, so the IP addresses are assigned automatically.

I will try with another Ethernet module because I cannot imagine what could be wrong.

Thanks you.

arangu97, 2021-02-23:
Hey, I have tried with another module and the WebServer with static IP is working!!

However the DHCP addressing is not working. 

_Trying to get an IP address using DHCP
Failed to configure Ethernet using DHCP
Local IP : 0.0.0.0
Subnet Mask : 0.0.0.0
Gateway IP : 0.0.0.0
DNS Server : 0.0.0.0_

Why this could be happening? Is the router not being able to detect the ESP32 or the hardware of the module?

Thanks.

RoSchmi, 2021-02-24:
Hi, I just tried your code on a Wio Terminal board and it is working.
Please include this code in Setup to see if the rigth SPI pins for your board are used.

void setup() {
  Serial.begin(115200);
  while (!Serial);

  //Ethernet.init(GPIO_NUM_5);  // ESP32 with Adafruit Featherwing Ethernet

   // Just info to know how to connect correctly
  Serial.println("=========================");
  Serial.println("Currently Used SPI pinout:");
  Serial.print("MOSI:");
  Serial.println(MOSI);
  Serial.print("MISO:");
  Serial.println(MISO);
  Serial.print("SCK:");
  Serial.println(SCK);
  Serial.print("SS:");
  Serial.println(SS);
  delay(2000);
  ....
  ....

If the pins are correct, I would try to use  10kOhm resistors as pull ups on the SPI lines.

jandrassy, 2021-02-24:
@RoSchmi, SPI already works for @arangu97

arangu97, 2021-02-26:
Yes, communication is working, but only when an static IP is assigned. 
Have you tried the DHCP example and is working for you?

RoSchmi, 2021-02-26:
Yes, on my Board (Wio Terminal = not ESP32) your example 'ethernet_test.zip' is working using DHCP. Or do you mean another example, then please give a link. If you use static IP with your 'ethernet_test.zip' example with the the command Ethernet.begin(mac, ip);, can you verify that the Enc28 is listed in the list of active devices in your router?
Have you tried to leave the command 'Ethernet.init(GPIO_NUM_5);'
On the page -https://github.com/khoih-prog/EthernetWebServer_SSL, there are many hints how to initialize different Ethernet modules on different boards

RoSchmi, 2021-02-26:
@arangu97 Just tried with an ESP32 board, DHCP works, router is a 'FritzBox'

[image]

arangu97, 2021-02-26:
Yes thank you, I have also tried with another router connected to Internet and DHCP is working.

Maybe the router without WAN connection(only for LAN testing) was not using properly the DHCP protocol....

closed 2021-02-26 by jandrassy

bkaraceylan, 2021-05-09:
Same problem here. Using ESP32 dev getting EthernetNoHardware, even though the UIPEthernet library works fine.

jandrassy, 2021-05-09:
> Same problem here. Using ESP32 dev getting EthernetNoHardware, even though the UIPEthernet library works fine.

do you have Ethernet.init to set the CS pin?

bkaraceylan, 2021-05-09:
Yes, I do have Ethernet.init(5) and CS is connected to GPIO5.

jandrassy, 2021-05-09:
reduce the SPI speed in utility/Enc28J60Network.cpp 
`#define SPI_ETHERNET_SETTINGS SPISettings(20000000, MSBFIRST, SPI_MODE0)`
to 10 000 000 or less

bkaraceylan, 2021-05-09:
10 000 000, 8 000 000, 4 000 000 didn't work how low should I try?

RoSchmi, 2021-05-13:
@bkaraceylan If you are still having problems, you could post your actual code.

bkaraceylan, 2021-05-15:
It is the basic example but sure here it is:

```
#include <EthernetENC.h>

EthernetClient ethClient;

uint8_t mac[6] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};

void BeginEth(){
    Serial.println("Init eth");
    
    if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
    // Check for Ethernet hardware present
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
      while (true) {
        delay(1); // do nothing, no point running without Ethernet hardware
      }
    } 
    if (Ethernet.linkStatus() == LinkOFF) {
      Serial.println("Ethernet cable is not connected.");
    }
    // try to configure using IP address instead of DHCP:
    //Ethernet.begin(mac, ip, myDns);
  } else {
    Serial.print("DHCP assigned IP ");
    Serial.println(Ethernet.localIP());
  }
}

void setup(){
  Serial.begin(115200);
  BeginEth();
}

void loop(){
}
```

RoSchmi, 2021-05-16:
@bkaraceylan No problems here with your code on an Esp32 Dev board and the latest EthernetEnc library. Got an Ip-Address by DHCP. Perhaps you should doublecheck your electrical connections.

jandrassy, 2021-05-16:
EthernetENC asks the DHCP to send the IP address as unicast because broadcast receiving is turned off (UIPEthernet has it on). if the DHCP server sends the UDP message with the IP lease as broadcast, it is not received.
try with static IP address.

VansonLeung, 2021-09-04:
I freaking swapped the MOSI and MISO wirings and then it suddenly worked. 
If your module does not respond, make sure you have tried it as well.