Networking-for-Arduino/EthernetENC #9: Not receiving DHCP Offer Packet

Craft4Cube, 2021-04-14:
I've just tried to use the EthernetENC library with the DHCP Address Printer example, and get the following output:

```
Initialize Ethernet with DHCP:
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
udp beginPacket, rip: 255.255.255.255, port: 67
udp, uip_poll preparing packet to send: 1, size: 325
udp, uip_packet to send: 0
Failed to configure Ethernet using DHCP
```

But I can see in Wireshark that the module is sending it's DHCP Discover, and the Router is responding with a DHCP Offer.

```
No.	Time	Source	Destination	Protocol	Length	Info
113	5.342738	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4a5
154	7.345124	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4a5
190	9.347130	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4a6
191	9.347631	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4a6
221	13.404441	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4a7
222	13.404441	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4a7
246	17.459337	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4a8
247	17.459797	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4a8
293	21.515242	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4a9
294	21.525351	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4a9
303	25.571324	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4aa
304	25.571758	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4aa
316	29.627390	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4ab
317	29.628571	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4ab
380	33.683590	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4ac
381	33.683994	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4ac
406	37.739906	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4ad
407	37.739906	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4ad
451	41.795579	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4ae
452	41.796027	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4ae
599	45.852460	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4af
600	45.852460	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4af
5270	49.908088	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4b0
5271	49.908334	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4b0
5368	53.964271	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4b1
5369	53.964271	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4b1
5457	58.020850	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4b2
5458	58.020850	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4b2
```

I also check in the DHCP.c file. It seems that `_dhcpUdpSocket.parsePacket()` in `DhcpClass::parseDHCPResponse` never get's a packet.

Craft4Cube, 2021-04-14:
I also just checked the same sketch with the UIPEthernet library. With it, it seems to work...

```
Initialize Ethernet with DHCP:
ENC28J60::init DEBUG:csPin = 42
ENC28J60::init DEBUG:miso = 44
ENC28J60::init DEBUG:mosi = 45
ENC28J60::init DEBUG:sck = 43
ENC28J60::init DEBUG:Use SPI lib SPI.begin()
ENC28J60::init INFO: Chip erevid=6 initialization completed.
DhcpClass::beginWithDHCP(uint8_t *mac) DEBUG_V1:Function started
DhcpClass::reset_DHCP_lease(void) DEBUG_V1:Function started
DhcpClass::request_DHCP_lease(void) DEBUG_V1:Function started
DhcpClass::presend_DHCP(void) DEBUG_V1:Function started (Empty function)
DhcpClass::request_DHCP_lease(void) DEBUG_V1:dhcp_state=STATE_DHCP_START -> send_DHCP_MESSAGE DHCP_DISCOVER
DhcpClass::send_DHCP_MESSAGE(uint8_t messageType, uint16_t secondsElapsed) DEBUG_V1:Function started
UIPUDP::beginPacket DEBUG:udp beginPacket, rip: 255.255.255.255, port: 67
DhcpClass::printByte(char * buf, uint8_t n ) DEBUG_V1:Function started
DhcpClass::printByte(char * buf, uint8_t n ) DEBUG_V1:Function started
DhcpClass::printByte(char * buf, uint8_t n ) DEBUG_V1:Function started
uipudp_appcall(void) DEBUG:udp, uip_poll preparing packet to send: 1, size: 325
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: CE02
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(1)[42-317]: 71B3
UIPEthernetClass::network_send() DEBUG:uip_packet: 1, hdrlen: 42
Enc28J60Network::sendPacket(memhandle handle) DEBUG:sendPacket(1) [801-93E]: 0 FF FF FF FF FF FF 0 AA BB CC DE 2 8 0 45 0 1 2F 0 1 0 0 40 11 79 BE 0 0 0 0 FF FF FF FF 0 44 0 43 1 1B 4C 8E 1 1 6 0 0 0 4 A4 0 0 80 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 AA BB CC DE 2 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 63 82 53 63 35 1 1 3D 7 1 0 AA BB CC DE 2 C C 45 4E 43 32 38 4A 43 43 44 45 30 32 37 6 1 3 6 F 3A 3B FF 
UIPUDP::_send() DEBUG:udp, uip_packet to send: 0
DhcpClass::parseDHCPResponse(uint32_t& transactionId) DEBUG_V1:Function started
Enc28J60Network::receivePacket(void) DEBUG:receivePacket [6-254], next: 258, stat: C0, Packet count: 1 -> OK
Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=590
UIPEthernetClass::tick() DEBUG:receivePacket: 255
UIPEthernetClass::tick() DEBUG:readPacket type IP, uip_len: 590
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: DC94
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(255)[42-590]: FFFF
uipudp_appcall(void) DEBUG:udp, uip_newdata received packet: 1, size: 548
UIPEthernetClass::tick() DEBUG:freeing packet: 255
Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:257
UIPUDP::parsePacket(void) DEBUG:udp parsePacket received packet: 1, size: 548
MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
DhcpClass::request_DHCP_lease(void) DEBUG_V1:dhcp_state=STATE_DHCP_DISCOVER,messageType=DHCP_OFFER -> send_DHCP_MESSAGE DHCP_REQUEST
DhcpClass::send_DHCP_MESSAGE(uint8_t messageType, uint16_t secondsElapsed) DEBUG_V1:Function started
UIPUDP::beginPacket DEBUG:udp beginPacket, rip: 255.255.255.255, port: 67
DhcpClass::printByte(char * buf, uint8_t n ) DEBUG_V1:Function started
DhcpClass::printByte(char * buf, uint8_t n ) DEBUG_V1:Function started
DhcpClass::printByte(char * buf, uint8_t n ) DEBUG_V1:Function started
uipudp_appcall(void) DEBUG:udp, uip_poll preparing packet to send: 1, size: 337
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: E602
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(1)[42-329]: 549F
UIPEthernetClass::network_send() DEBUG:uip_packet: 1, hdrlen: 42
Enc28J60Network::sendPacket(memhandle handle) DEBUG:sendPacket(1) [801-94A]: 0 FF FF FF FF FF FF 0 AA BB CC DE 2 8 0 45 0 1 3B 0 2 0 0 40 11 79 B1 0 0 0 0 FF FF FF FF 0 44 0 43 1 27 60 AB 1 1 6 0 0 0 4 A4 0 0 80 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 AA BB CC DE 2 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 63 82 53 63 35 1 3 3D 7 1 0 AA BB CC DE 2 C C 45 4E 43 32 38 4A 43 43 44 45 30 32 32 4 C0 A8 0 70 36 4 C0 A8 0 1 37 6 1 3 6 F 3A 3B FF 
UIPUDP::_send() DEBUG:udp, uip_packet to send: 0
DhcpClass::parseDHCPResponse(uint32_t& transactionId) DEBUG_V1:Function started
Enc28J60Network::receivePacket(void) DEBUG:receivePacket [25E-2B4], next: 2B8, stat: C0, Packet count: 2 -> OK
Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=86
UIPEthernetClass::tick() DEBUG:receivePacket: 255
UIPEthernetClass::tick() DEBUG:readPacket type IP, uip_len: 86
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: 7D1A
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(255)[42-86]: FFFF
UIPEthernetClass::tick() DEBUG:freeing packet: 255
Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:2B7
[... cut ...]
DhcpClass::getSubnetMask(void) DEBUG_V1:Function started
My IP address: 192.168.0.112
```

```
No.	Time	Source	Destination	Protocol	Length	Info
17499	444.634012	0.0.0.0	255.255.255.255	DHCP	317	DHCP Discover - Transaction ID 0x4a4
17500	444.634462	192.168.0.1	255.255.255.255	DHCP	590	DHCP Offer    - Transaction ID 0x4a4
17502	444.955735	0.0.0.0	255.255.255.255	DHCP	329	DHCP Request  - Transaction ID 0x4a4
17503	444.955735	192.168.0.1	255.255.255.255	DHCP	590	DHCP ACK      - Transaction ID 0x4a4
```

jandrassy, 2021-04-14:
yes. EthernetENC requests the DHCP response to be send as unicast, not as broadcast, because broadcasts are filter out (except of ARP).
https://github.com/jandrassy/EthernetENC/wiki/Limitations

Craft4Cube, 2021-04-14:
Ah I see. Adding `| ERXFCON_BCEN` makes it work indeed. Can this be added as an define option maybe?

jandrassy, 2021-04-14:
> 
> 
> Ah I see. Adding `| ERXFCON_BCEN` makes it work indeed. Can this be added as an define option maybe?

this puts constant load on the memory and processing. every broadcast packet on network  is stored in RX buffer and processed. and all this for one DHCP packet.

Craft4Cube, 2021-04-14:
Would there be a way to enable Broadcast only when needed?

jandrassy, 2021-04-14:
why does you DHCP server send the response as broadcast? for me the unicast didn't work only when I used an Ethernet to WiFi adapter. 
 
> Would there be a way to enable Broadcast only when needed? Like, enable it just before sending DHCP and disabeling it afterwards?

You could try it. The datasheet doesn't constrain when the filter can be set. But I didn't test. I didn't put time into this since I didn't know how common is the issue.

Craft4Cube, 2021-04-14:
> why does you DHCP server send the response as broadcast? for me the unicast didn't work only when I used an Ethernet to WiFi adapter.

I don't know. I'm using a Huawei Router in it's default config.

> You could try it. The datasheet doesn't constrain when the filter can be set. But I didn't test. I didn't put time into this since I didn't know how common is the issue.

I'll try when I find some time. Since the router I'm using is the one provided by my ISP, I think the issue might be more common, than expected.

nzy, 2021-04-27:
> `ERXFCON_BCEN`

Where do you added this peace of code?

RaysoftTechnologiesInc, 2022-03-17:
hi @Craft4Cube,

I am working on interfacing the ENC with a nucleo using mbed, I am currently stuck because its not working, using DHCP, would you please share with me exactly where I need to edit in the code so that DHCP can work?

Craft4Cube, 2022-03-21:
For anyone interested in the change I made, you can find it here: https://github.com/NetCube-Systems-Austria/EthernetENC/commit/faaf3f2b4ff744fcec7ec02d16199e1e8d907ed1