Networking-for-Arduino/EthernetENC #10: Arduino Mega or Mega 2560

markoshenoda, 2021-04-25:
Can't compile for Mega 2560

jandrassy, 2021-04-25:
I can

markoshenoda, 2021-04-25:
> Executing task in folder ArdouinoSQL: C:\Users\AnyApp\.platformio\penv\Scripts\platformio.exe run <

Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.3.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <MySQL_MariaDB_Generic> 1.0.3
|   |-- <EthernetENC> 2.0.0
|-- <EthernetENC> 2.0.0
Building in release mode
Compiling .pio\build\megaatmega2560\src\main.cpp.o
Compiling .pio\build\megaatmega2560\libac9\EthernetENC\Dhcp.cpp.o
Compiling .pio\build\megaatmega2560\libac9\EthernetENC\Dns.cpp.o
Compiling .pio\build\megaatmega2560\libac9\EthernetENC\Ethernet.cpp.o
Compiling .pio\build\megaatmega2560\libac9\EthernetENC\EthernetClient.cpp.o
Compiling .pio\build\megaatmega2560\libac9\EthernetENC\EthernetServer.cpp.o
Compiling .pio\build\megaatmega2560\libac9\EthernetENC\EthernetUdp.cpp.o
Compiling .pio\build\megaatmega2560\libac9\EthernetENC\utility\Enc28J60Network.cpp.o
Compiling .pio\build\megaatmega2560\libac9\EthernetENC\utility\mempool.cpp.o
Compiling .pio\build\megaatmega2560\libac9\EthernetENC\utility\uip.c.o
Compiling .pio\build\megaatmega2560\libac9\EthernetENC\utility\uip_arp.c.o
Archiving .pio\build\megaatmega2560\libc30\libMySQL_MariaDB_Generic.a
.pio\libdeps\megaatmega2560\EthernetENC\src\utility\Enc28J60Network.cpp:27:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

 #include <SPI.h>
Archiving .pio\build\megaatmega2560\libFrameworkArduinoVariant.a
          ^~~~~~~
compilation terminated.
Compiling .pio\build\megaatmega2560\FrameworkArduino\CDC.cpp.o
*** [.pio\build\megaatmega2560\libac9\EthernetENC\utility\Enc28J60Network.cpp.o] Error 1
==================================================================== [FAILED] Took 0.77 seconds ====================================================================
The terminal process "C:\Users\AnyApp\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

jandrassy, 2021-04-25:
and if you add the SPI library to the project?

markoshenoda, 2021-04-25:
i'm srry I’m a newbie in this how to that

jandrassy, 2021-04-25:
> i'm srry I’m a newbie in this how to that

use Arduino IDE. platformio is not Arduino and it is not for newbies

markoshenoda, 2021-04-25:
it work's thx, i find the fix 

srry for the troubles