Networking-for-Arduino/EthernetENC #11: Olimexino-328 and MOD-ENC28J60 Ethernet not found

gregor-horvath, 2021-05-15:
Should this library work with the Hardware ?:
https://www.olimex.com/Products/Duino/AVR/OLIMEXINO-328/open-source-hardware
https://www.olimex.com/Products/Modules/Ethernet/MOD-ENC28J60/

Hardware works with https://www.aelius.com/njh/ethercard/index.html

Serial output of the example:

> Ethernet Example
Ethernet shield was not found.  Sorry, can't run without hardware. :(

Example sketch code:

`
#include <SPI.h>
#include <EthernetENC.h>

// network configuration.  gateway and subnet are optional.

 // the media access control (ethernet hardware) address for the shield:
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };  
//the IP address for the shield:
byte ip[] = { 192, 168, 0, 77 };    
// the router's gateway address:
byte gateway[] = { 192, 168, 0, 1 };
// the subnet:
byte subnet[] = { 255, 255, 255, 0 };

// telnet defaults to port 23
EthernetServerPrint server = EthernetServerPrint(23);

void setup()
{
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.println("Ethernet Example");
  // initialize the ethernet device
  Ethernet.begin(mac, ip, gateway, subnet);
 // Check for Ethernet hardware present
  if (Ethernet.hardwareStatus() == EthernetNoHardware) {
    Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
    while (true) {
      delay(1); // do nothing, no point running without Ethernet hardware
    }
  }
  // start listening for clients
  server.begin();
}

void loop()
{
  // if an incoming client connects, there will be bytes available to read:
  EthernetClient client = server.available();
  if (client == true) {
    // read bytes from the incoming client and write them back
    // to any clients connected to the server:
    server.write(client.read());
  }
}
`

jandrassy, 2021-05-15:
try to set the CS pin with Ethernet.init

gregor-horvath, 2021-05-15:
I added 
` Ethernet.init(7); `
and it worked! Thank you a lot!

I also noticed that there is a hardware jumper on the Olimexino-328 to set the UEXT_CS pin.

closed 2021-05-15 by jandrassy