Networking-for-Arduino/EthernetENC #15: UDP services, does not work with ESP8266 arduino core above 2.7.4

NourbakhshRad, 2021-07-17:
I am trying to use UIPEthernet with ESP8266 arduino core 3.0.1, my hardware is ESP-12E and ENC28J60 Ethernet shield
Unfortunately UDP services like DHCP, SNMP etc. are not working properly and I had to downgrade my ESP8266 arduino core to 2.7.4
I also tried to change the definitions of conflict but to no avail
Any suggestion?! Thanks

jandrassy, 2021-07-17:
revert this in your copy
https://github.com/esp8266/Arduino/commit/08f170510d69423ab58b0fa1edf31b4a5373eba4#diff-2d82230ef328e5fa01af7ef0363e28f757f61016abf8fbb5de06827ec6a006aa

NourbakhshRad, 2021-07-18:
Thanks @jandrassy
With this change, my problem was solved!

jandrassy, 2021-07-18:
EthernetUdp.cpp had a haphazardn ip address check, which resulted in this issue after the change in IPAddress in the esp8266 core. 
I fixed the check in EthernetUdp.

closed 2021-07-18 by jandrassy