Networking-for-Arduino/EthernetENC #20: Soft Reset of the ESP8266

rajeshdoshi, 2021-10-16:
Hi,
I have connected ESP8266 to ENC28J60. Using 5 and 4 pins for CS and Reset respectively. 
Following code works beautifully well. 

#include <EthernetENC.h>

typedef unsigned char N8;

#define LED D0

EthernetServer server(port);
uint8_t mac[6] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
IPAddress ip = {192,168,131,171};
uint8_t myDns[] = {192,168,131,1};

unsigned long ledmil;
unsigned char ledon=0;
unsigned char xmbuf[60];
unsigned char datrdy=0;
int xmcnt;

//=======================================================================
//                    Power on setup
//=======================================================================
void setup() 
{
  Serial.begin(9600);
  Serial.println();

  pinMode(4, OUTPUT); 
  digitalWrite(4,LOW);
  delay(100);
  digitalWrite(4,HIGH);

  Ethernet.init(5);
  Ethernet.begin(mac,ip,myDns); //Connect to wifi

  Serial.println(Ethernet.localIP());
  server.begin();
  ledmil=millis();
}
//=======================================================================
//                    Loop
//=======================================================================
unsigned char buf;
unsigned long curmil;

void loop() 
{
  EthernetClient client = server.available();
  
  if (client) {
    if(client.connected())
    {
      Serial.println("Client Connected");
    }
    
    while(client.connected()){      
      while(client.available()>0){
        // read data from the connected client
        buf=client.read();
        Serial.write(buf); 
        if(buf=='X')
          { sprintf((char *)xmbuf,"Hello World1"); datrdy=1; }
        else if(buf=='B')
          { sprintf((char *)xmbuf,"Hello World2"); datrdy=1; }
        Ethernet.maintain();
      }
      //Send Data to connected client
      while(Serial.available()>0)
     {
        client.write(Serial.read());
      }
   }
    client.stop();
    Serial.println("Client disconnected");    
  Ethernet.maintain();
  }
  Ethernet.maintain();
}//=======================================================================

However, moment I replace
      while(Serial.available()>0)
     {
        client.write(Serial.read());
      }

with 
    if(datrdy)
     {
        client.write (xmbuf,13');
        datrdy=0;
     }
ESP8266 starts resetting randomly on receipt of data. 

<LF>
Soft WDT reset<CR><LF>
<CR><LF>
>>>stack>>><CR><LF>
<CR><LF>
ctx: cont<CR><LF>
sp: 3ffffd10 end: 3fffffc0 offset: 01a0<CR><LF>
3ffffeb0:  3ffe85d9 3ffee852 00000000 40100298  <CR><LF>
3ffffec0:  3ffe85d9 3ffee852 39000000 40203bcf  <CR><LF>
3ffffed0:  3ffee852 3fffff80 3ffee852 3ffee8f8  <CR><LF>
3ffffee0:  3ffee852 3fffff80 00000039 40203d60  <CR><LF>
3ffffef0:  00000000 00000000 3ffee5a7 4020420e  <CR><LF>
3fffff00:  00001d83 3ffe0001 40100209 00000000  <CR><LF>
3fffff10:  00001f84 3fffff60 3fffff80 3ffee8f8  <CR><LF>
3fffff20:  3fffdad0 3ffee538 3ffee5a7 3ffee8f8  <CR><LF>
3fffff30:  3fffdad0 3fffff80 3ffee5a7 402028cd  <CR><LF>
3fffff40:  3fffdad0 3ffee538 3ffee539 3ffee8f8  <CR><LF>
3fffff50:  3fffdad0 3fffff80 3fffff80 40202d8c  <CR><LF>
3fffff60:  3fffdad0 0000000a 3ffee539 40202f1e  <CR><LF>
3fffff70:  3fffdad0 3ffee538 3ffee539 40202717  <CR><LF>
3fffff80:  40208f54 00000000 000003e8 40100168  <CR><LF>
3fffff90:  00000000 3ffee5aa 3ffee8e4 40100189  <CR><LF>
3fffffa0:  feefeffe 00000000 3ffee8e4 402062ec  <CR><LF>
3fffffb0:  feefeffe feefeffe 3ffe85ec 40100bc9  <CR><LF>
<<<stack<<<<CR><LF>
<LF>
--------------- CUT HERE FOR EXCEPTION DECODER ---------------<LF>

I am clueless about it as of now.

jandrassy, 2021-10-16:
use the stack decoder IDE plugin
https://github.com/me-no-dev/EspExceptionDecoder

rajeshdoshi, 2021-10-17:
Hi. 
Thanks for your reply and help. 
I have found that it is the delay which makes the difference. If I do not have delay called, the ESP resets abruptly. 
I have no explanation for it.

closed 2022-02-09 by jandrassy