Networking-for-Arduino/EthernetENC #21: ENC28J60 and SD card module on ESP32

Ganesh-Shanbhag-face, 2021-11-16:
Hi, I am using ENC28J60 Ethernet module (SPI) and SD Card module (SPI) with ESP32 as I need wired networking on ESP32. Ethernet Module and SD Card module are using different CS pin. Ethernet Module when connected to ESP32 without connecting the SD Card module will get the IP from DHCP. But when connected with SD Card module won't work but the SD Card module works alone. What may be the problem?

And when used Ethernet Module alone although it gets IP from DHCP is not connecting to MQTT (Hive MQ Public Client). What may be the problem?

jandrassy, 2021-11-16:
I don't recommend to use this library with the esp32. esp32 as much better options for Ethernet.

The problem could be SPI speed. Try to set lower speed than the 20MHz here
https://github.com/jandrassy/EthernetENC/blob/c735fd4cf6821a166ef6b36dd56bd861ba61c400/src/utility/Enc28J60Network.cpp#L39

Ganesh-Shanbhag-face, 2021-11-17:
@jandrassy I tried with that but it did not work. Can you suggest any module which works with this library on ESP32?

jandrassy, 2021-11-17:
do you use a 3.3 V powered module or a module with 5 V powering and a level shifter chip?

Ganesh-Shanbhag-face, 2021-11-17:
Ethernet module is working on 3.3V and SD Card module with 5V with inbuilt 3.3V regulator. So, literally both are working on 3.3V.

jandrassy, 2021-11-17:
but then the SD card module should convert 5 V SPI to 3.3 V SPI and converts 3.3 V SPI to ?.? V SPI

Ganesh-Shanbhag-face, 2021-11-17:
No for SD Card module there is inbuilt regulator which will convert it into 3.3V and supplies to the controller/buffer as well as to the SD Card. Since the controller/buffer and SD card are powered by 3.3V there levels are 3.3V compatible.

jandrassy, 2021-11-17:
which side of the 'buffer'?

Ganesh-Shanbhag-face, 2021-11-17:
SD Card module is using LVC125A.

jandrassy, 2021-11-17:
> LVC125A

then it is OK

simogaspa84, 2022-02-08:
> I don't recommend to use this library with the esp32. esp32 as much better options for Ethernet.
> 
> The problem could be SPI speed. Try to set lower speed than the 20MHz here
> 
> https://github.com/jandrassy/EthernetENC/blob/c735fd4cf6821a166ef6b36dd56bd861ba61c400/src/utility/Enc28J60Network.cpp#L39

Does it work this library with ESP32 ? Thanks a lot

jandrassy, 2022-02-08:
> Does it work this library with ESP32 ? Thanks a lot

It does work even it is not a recommended Ethernet solution for esp32

closed 2022-02-08 by jandrassy

jandrassy, 2022-02-09:
> Thanks a lot for the quick reply... Can I ask you why it is not recommended ? Thanks a lot

esp32 has LwIP TCP/IP stack which handles WiFi, SoftAP and the esp32 Ethernet peripheral.
EthernetENC uses uIP TCP/IP stack and was designed for small MCU with a small RAM.