Networking-for-Arduino/EthernetENC #22: 'class EthernetServer' has no member named 'write'

matkor, 2022-01-16:
I am trying to replace UIPEthernet with EthernetENC for my tiny MySensors project but compilation fails with
```
/home/matkor/Arduino/libraries/MySensors/core/MyGatewayTransportEthernet.cpp: In function 'bool gatewayTransportSend(MyMessage&)':
/home/matkor/Arduino/libraries/MySensors/core/MyGatewayTransportEthernet.cpp:297:27: error: 'class EthernetServer' has no member named 'write'
  nbytes = _ethernetServer.write(_ethernetMessage);
                           ^~~~~
Using library EthernetENC at version 2.0.2 in folder: /home/matkor/Arduino/libraries/EthernetENC 
Using library MySensors at version 2.3.2 in folder: /home/matkor/Arduino/libraries/MySensors 
```

Is it my error, MySensors or EthernetENC just lacks that method now?
Regards, TIA

ondras12345, 2022-01-16:
https://github.com/jandrassy/EthernetENC/wiki/Examples
> In PagerServer, ChatServer and DhcpChatServer examples please use EthernetServerPrint server(23). EthernetServer in EthernetENC library doesn't support writing to all clients because it takes flash and RAM and has little use. This function is only in EthernetServerPrint class.

jandrassy, 2022-01-16:
and https://github.com/jandrassy/EthernetENC/wiki/Limitations

closed 2022-02-09 by jandrassy