Networking-for-Arduino/EthernetENC #25: Setting all PIN in ENC28J60 shield

simogaspa84, 2022-02-10:
Hi... I am trying this example to use your library with ESP32 and ENC28J60 shield.

``
#include <SPI.h>
#include <EthernetENC.h>

// Enter a MAC address and IP address for your controller below.
// The IP address will be dependent on your local network:
byte mac[] = {
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED
};
IPAddress ip(192, 168, 1, 177);

// Initialize the Ethernet server library
// with the IP address and port you want to use
// (port 80 is default for HTTP):

EthernetServer server_ethernet(80);
EthernetClient client_ethernet;

// Variables to measure the speed
unsigned long beginMicros, endMicros;
unsigned long byteCount = 0;
bool printWebData = true;  // set to false for better speed measurement

void Ethernet_Init(void)
{

  // You can use Ethernet.init(pin) to configure the CS pin
  Ethernet.init(15);  // Most Arduino shields

  // Open serial communications and wait for port to open:
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.println("Ethernet WebServer Example");

  // start the Ethernet connection and the server:
  Ethernet.begin(mac, ip);

  // Check for Ethernet hardware present
  if (Ethernet.hardwareStatus() == EthernetNoHardware) 
  {
    Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
    while (true) 
    {
      delay(1); // do nothing, no point running without Ethernet hardware
    }
  }
  if (Ethernet.linkStatus() == LinkOFF) 
  {
    Serial.println("Ethernet cable is not connected.");
  }

  // start the server
  server_ethernet.begin();
  Serial.print("server is at ");
  Serial.println(Ethernet.localIP());

}

void Ethernet_Task(void)
{   
     // if there are incoming bytes available
  // from the server, read them and print them:
  int len = client_ethernet.available();
  if (len > 0) {
    byte buffer[80];
    if (len > 80) len = 80;
    client_ethernet.read(buffer, len);
    if (printWebData) {
      Serial.write(buffer, len); // show in the serial monitor (slows some boards)
    }
    byteCount = byteCount + len;
  }

  // if the server's disconnected, stop the client:
  if (!client_ethernet.connected()) {
    endMicros = micros();
    Serial.println();
    Serial.println("disconnecting.");
    client_ethernet.stop();
    Serial.print("Received ");
    Serial.print(byteCount);
    Serial.print(" bytes in ");
    float seconds = (float)(endMicros - beginMicros) / 1000000.0;
    Serial.print(seconds, 4);
    float rate = (float)byteCount / seconds / 1000.0;
    Serial.print(", rate = ");
    Serial.print(rate);
    Serial.print(" kbytes/second");
    Serial.println();

    // do nothing forevermore:
    while (true) {
      delay(1);
    }
  }
``
I got always the same message 
Ethernet shield was not found.  Sorry, can't run without hardware . 

The point is that I have the following HW configuration 

[image]

Does the library provide a way for specify my special pinout ? 

I just saw that you can select the CS PIn,,,, 

Thanks a lot for your huge support...

jandrassy, 2022-02-11:
esp32 arduino doesn't set the SPI frequency to value it can handle. the library sets 20 MHz and expects the underlying code to limit it to a value it can handle.
set a lower frequency in Enc28J60Network in SPISettings

simogaspa84, 2022-02-12:
Hi jandrassy... I didn't understand if with my special pinout your code should work.. 
Does the library work with all possible pinout setting?
Do I have to change something and how ? 

the freuency I think I should change this right ? 

#define SPI_ETHERNET_SETTINGS SPISettings(spiClk, MSBFIRST, SPI_MODE0)

I am trying this example but I always have  issue 

``
#include "ethernet_module.h"
#include <SPI.h>
#include <EthernetENC.h>

// Enter a MAC address and IP address for your controller below.
// The IP address will be dependent on your local network:
byte mac[] = {
  0x02, 0x00, 0x00, 0x12, 0x34, 0x56
};
IPAddress ip(192, 168, 0, 35);

// Initialize the Ethernet server library
// with the IP address and port you want to use
// (port 80 is default for HTTP):

EthernetServer server_ethernet(80);
EthernetClient client_ethernet;

// Variables to measure the speed
unsigned long beginMicros, endMicros;
unsigned long byteCount = 0;
bool printWebData = true;  // set to false for better speed measurement

void Ethernet_Init(void)
{

  // You can use Ethernet.init(pin) to configure the CS pin
  Ethernet.init(15);  // Most Arduino shields

  // Open serial communications and wait for port to open:
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.println("Ethernet WebServer Example");

  // start the Ethernet connection and the server:
  Ethernet.begin(mac, ip);

  // Check for Ethernet hardware present
  if (Ethernet.hardwareStatus() == EthernetNoHardware) 
  {
    Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
    while (true) 
    {
      delay(1); // do nothing, no point running without Ethernet hardware
    }
  }
  if (Ethernet.linkStatus() == LinkOFF) 
  {
    Serial.println("Ethernet cable is not connected.");
  }

  // start the server
  server_ethernet.begin();
  Serial.print("server is at ");
  Serial.println(Ethernet.localIP());

}
``

simogaspa84, 2022-02-12:
The point is Do I have to change the following code for my hw setup and how ? 

void Enc28J60Network::initSPI()
{
  if (spiInitialized)
    return;
  pinMode(csPin, OUTPUT);
  CSPASSIVE;
  SPI.begin();
  spiInitialized = true;
}

jandrassy, 2022-02-12:
> The point is Do I have to change the following code for my hw setup and how ?

SPI library is used without any pin settings, so I guess only default pins will work. Use the default SPI pins.

simogaspa84, 2022-02-12:
Mr jandrassy.. 
you deserve really a coffee for your work..
 But I think we must provide a library which is fully configurable because we cannot force the hw wiring.
I cannot use the default SPI pins because hw is aready done .
Maybe I found where there is the issue but I need your support to confirm it. 
If we look at this code 

void Enc28J60Network::initSPI()
{

  if (spiInitialized)
    return;
  pinMode(csPin, OUTPUT);
  CSPASSIVE;
  SPI.begin();
  spiInitialized = true;
}

We must check how and where the object SPI is defined.. 
I found that it is defined here   (SPI.cpp file)

void SPIClass::writePattern_(const uint8_t * data, uint8_t size, uint8_t repeat)
{
    uint8_t bytes = (size * repeat);
    uint8_t buffer[64];
    uint8_t * bufferPtr = &buffer[0];
    const uint8_t * dataPtr;
    uint8_t dataSize = bytes;
    for(uint8_t i = 0; i < repeat; i++) {
        dataSize = size;
        dataPtr = data;
        while(dataSize--) {
            *bufferPtr = *dataPtr;
            dataPtr++;
            bufferPtr++;
        }
    }

    writeBytes(&buffer[0], bytes);
}

**SPIClass SPI(VSPI);**

I think that we are not able to change here my hw configuration never works... 
I should set HSPI and then maybe running other configuration for example 

 SPI.begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_SS);

What do you think ? 

Thanks again

jandrassy, 2022-02-12:
I think, if you call SPI.begin() with the pin numbers in sketch, then the SPI,begin() call without parameters in EthernetENC will do nothing and SPI will use your pins.

jandrassy, 2022-02-12:
direct Ethernet connection between two devices is not as simple as inserting cable into a socket. 
did you configure the network on both sides? do you use a crossed cable?

simogaspa84, 2022-02-12:
Ok so lt's pu it in an esay way.. 
Can you describe the setup in order to test this example 

https://github.com/arduino-libraries/Ethernet/blob/master/examples/WebServer/WebServer.ino

Thanks a lot

jandrassy, 2022-02-13:
try the WiFiClient example first

jamesgohl, 2022-07-18:
> > The point is Do I have to change the following code for my hw setup and how ?
> 
> SPI library is used without any pin settings, so I guess only default pins will work. Use the default SPI pins.

@JAndrassy simogaspa84 has a very good point about the CS pin of the SPI protocol, some devices have a defined set of hardware pins, the ATmega2650 is different to the arduino UNO for instance, but more importantly is that the CS pin is selectable, hence the term Cable Select, allowing multiple SPI devices to be used simulatenously with unique CS pin assignments.

I have not come across a library for driving an SPI device which does not allow you to pass in a CS pin value from outside the library, and this is usually passed in at library class instantiation, for example:

`EthernetServer server = EthernetServer(CS_PIN, LISTENPORT);`

I've created a constructor override to achieve this but, really it should come out of the box in my opinion.

JAndrassy, 2022-07-18:
@jamesgohl what is wrong with Ethernet.init(CS)?
https://www.arduino.cc/reference/en/libraries/ethernet/ethernet.init/