Networking-for-Arduino/EthernetENC #34: Enc28j60 Hangs or freeze

inationit, 2022-05-13:
Hi, i am a newbee...
i have been trying to connect ENC28j60 with ESP 32s .. CS Pin 14. 
i m able to communicate with my HMI with IP 192.168.1.100 and getting modbus tcp replies from it..
after running for 7 hours it freezes while sending the modbus request data to the HMI.
currently i am using WDT to reset if the system if it is not responding for more than 2 minutes...
i am trying to poll for every one minute... no memory leaks... 
here is my code..  (removed the WDT part)..
#include <EthernetENC.h>
#include <SPI.h>

//modbus tcp
const int MODBUS_CONNECT_ERROR = -10;
const int MODBUS_NO_RESPONSE = -11;
const byte FNC_READ_REGS = 0x03;
const byte FNC_WRITE_SINGLE = 0x06;
const byte FNC_ERR_FLAG = 0x80;
byte mac[] = {0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02};
const int ETH_CS_PIN = 14;
const IPAddress isgAddress(192, 168, 1, 100);
EthernetClient client;
int clientstatus =  0;
IPAddress ip(192,168,1,10);
IPAddress gateway(192,168,1,1);
IPAddress DNSServer(192,168,1,1);
IPAddress subnet(255, 255, 255, 0);
String hostname = "ESP32";
int rst_Ethernet = 0;
int client_value = 0;

void setup() {
  Serial.begin(115200);
  SPI.begin();
  pinMode(14,OUTPUT);
  digitalWrite(14,LOW);   
  delay(2000);   
  Serial.println("Configuring Ethernet"); 
  pinMode(4,OUTPUT);
  digitalWrite(4,LOW);
  delay(100);
  digitalWrite(4,HIGH);
  //pinMode(4,INPUT);
  delay(3000);  
  Ethernet.init(ETH_CS_PIN);
  Ethernet.begin(mac, ip, gateway, DNSServer, subnet);
  ethernet_connect();
}
  
int tests(Client& client) {
    short regs[16];
    int res = modbusRequest(client, 1, 0, 8, regs);
    if (res != 0)
    return res;
  return 0;
}

void ethernet_connect(){  
   client.connect(isgAddress, 8888);
   while(!client.connected()){
    client.connect(isgAddress, 8888);
    Serial.println("Trying to Connect TCP...");
    delay(3000);
   } 
    if(client.connected()){
    client.setTimeout(1000);
    Serial.println("client connected");
    clientstatus = 0;
   }else {
    
    Serial.println("connection failed");
  }

}

int modbusRequest(Client& client, byte uid, unsigned int addr, byte len, short *regs) {
  const byte CODE_IX = 7;
  const byte ERR_CODE_IX = 8;
  const byte LENGTH_IX = 8;
  const byte DATA_IX = 9;
  
   byte request[] = {0, 1, 0, 0, 0, 6, uid, FNC_READ_REGS, (byte) (addr / 256), (byte) (addr % 256), 0, len};
   Serial.println("Request");
   String bt_data = "";
   for(int i = 0; i < sizeof(request); ++i){
   Serial.print(request[i], HEX);
   Serial.print(' '); 
   bt_data += String(request[i], HEX);       
  }
  Serial.println("");
  
  unsigned long Eth_currentMillis = millis(); // store the current time
    
  client.write(request, sizeof(request));   
  int respDataLen = (len * 2);
  byte response[max((int) DATA_IX, respDataLen)];
  Serial.println("Response");
  int readLen = client.readBytes(response, DATA_IX);
  if (readLen < DATA_IX) {
     return MODBUS_NO_RESPONSE;
  }
  switch (response[CODE_IX]) {
    case FNC_READ_REGS:
      break;
    case (FNC_ERR_FLAG | FNC_READ_REGS):
      return response[ERR_CODE_IX]; // 0x01, 0x02, 0x03 or 0x11
    default:
      return -3;
  }
  readLen = client.readBytes(response, respDataLen);
       String br_data = "";
       for(int i = 0; i < respDataLen; ++i){
       Serial.print(response[i], HEX);
       Serial.print(' '); 
       br_data += String(response[i], HEX);       
      }  
   return 0;
}
void loop() {
  Ethernet.maintain();      
  int res = tests(client);
            if (res != 0) {
          Serial.print("modbus error ");
          Serial.println(res);
          }  
   Serial.println(ESP.getFreeHeap()); 
   delay(100);
   }

inationit, 2022-05-13:
i have downloaded the master version (this version) of yours  from github and implemented in my device.. will let you know if the issues still persists or not.. thanks a lot for a quick reply....

inationit, 2022-05-15:
Hi it seems issue got solved…. It’s been running for more than 42 hours now… thanks for your support …

closed 2022-05-15 by inationit

JAndrassy, 2022-05-15:
thank you for the feedback.

JAndrassy, 2022-05-15:
version 2.0.3 with the fix should be available in Library Manager in a few hours