Networking-for-Arduino/EthernetENC #35: Network connection goes deaf.

pvrfr, 2022-05-23:
Hello;

I am having the same problem with EthernetENC as I did with UIPEthernet.   That is the
stack stops receiving packets.   You fixed the problem and afterwards the code ran without
problems.   Perhaps there is a similar bug in your new code.   The processor is an ESP-12N.
The code is part of the OpenSprinkler project.

The issue was previously discussed here:
https://github.com/UIPEthernet/UIPEthernet/issues/129

JAndrassy, 2022-05-23:
the error I found 'hidden in plain sight' was not an error, but my confusion about `if (*this)` (which executes the bool operator). I replaced it with `if (data)` and after I realized my misunderstanding of `if (*this)` I changed it back.

```
UIPClient::operator bool()
{
  UIPEthernetClass::tick();
  return data && (!(data->state & UIP_CLIENT_REMOTECLOSED) || data->packets_in[0] != NOBLOCK);
}
```
`if (*this)` is used to skip available(), peek(), read() and discardReceived() (which is flush() n UIPEthernet).

JAndrassy, 2022-05-24:
the difference is `tick()` and this part of the bool operator implementation: 
`(!(data->state & UIP_CLIENT_REMOTECLOSED) || data->packets_in[0] != NOBLOCK)`
instead of `*this` in available(), peek(), read() try
`data && (!(data->state & UIP_CLIENT_REMOTECLOSED) || data->packets_in[0] != NOBLOCK)`
this will remove `tick()` from available(), peek(), read()