Networking-for-Arduino/EthernetENC #37: Limitation of length in UDP ?

rajeshdoshi, 2022-06-03:
I am finding that any packet more than 18 bytes is not being received by following code. 

#include <SPI.h>        
#include <EthernetENC.h>

// Enter a MAC address and IP address for your controller below.
// The IP address will be dependent on your local network:
byte mac[] = {  
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
IPAddress ip(192, 168, 1, 177);

unsigned int localPort = 35350;      // local port to listen on

// An EthernetUDP instance to let us send and receive packets over UDP
EthernetUDP Udp;

char packetBuffer[1024]; //buffer to hold incoming packet,

void setup() {
  // start the Ethernet and UDP:
  Serial.begin(115200);
  Ethernet.init(2);
  Ethernet.begin(mac,ip);
  Serial.println(Ethernet.localIP());
  Udp.begin(localPort);
  
}

void loop() {
  while(!Serial.available());
  int packetSize = Udp.parsePacket();
  if(Udp.available())
  {
    Serial.print("Received packet of size ");
    Serial.println(packetSize);
    Serial.print("From ");
    IPAddress remote = Udp.remoteIP();
    for (int i =0; i < 4; i++)
    {
      Serial.print(remote[i], DEC);
      if (i < 3)
      {
        Serial.print(".");
      }
    }
    Serial.print(", port ");
    Serial.println(Udp.remotePort());

    // read the packet into packetBuffer
    Udp.read(packetBuffer,255);
    Serial.println("Contents:");
    Serial.println(packetBuffer);
 }
}

rajeshdoshi, 2022-06-03:
I am using ESP32 connected to ENC28J60.

JAndrassy, 2022-06-03:
try to reduce the SPI speed, in SPI_ETHERNET_SETTINGS in Enc28J60Network.cpp

btw: IPAddress is printable. `Serial.print(remote);`

rajeshdoshi, 2022-06-04:
I changed to 
#define SPI_ETHERNET_SETTINGS SPISettings(10000000, MSBFIRST, SPI_MODE0)
then to
#define SPI_ETHERNET_SETTINGS SPISettings(4000000, MSBFIRST, SPI_MODE0)

There is no change. It still receives only 18 bytes.

JAndrassy, 2022-06-04:
how do you send the UDP message?

rajeshdoshi, 2022-06-04:
Using PacketSender

rajeshdoshi, 2022-06-04:
The observation is, if I broadcast a message from PacketSender, the message is received up to 18 characters. However, if it is directly addressed to the device IP address, then there is no such restriction.

JAndrassy, 2022-06-05:
broadcast shouldn't be received at all

rajeshdoshi, 2022-06-06:
I reinstalled the library. Still the observation is same. 18 bytes of broadcasted data is received.