Networking-for-Arduino/EthernetENC #38: Sending UDP message Error

rajeshdoshi, 2022-06-04:
Is it possible to broadcast a message on UDP ? I tried with 255.255.255.255 as IP address in UDP.beginPacket. 
It does return 1, however, message is not sent on the network. 

I can not IP address with only last digits as 255 because when device is connected on a network, the IP address of the PC is not known.

JAndrassy, 2022-06-04:
yes, it should work. how do you know it is not sent?

rajeshdoshi, 2022-06-04:
I am not receiving any data on PacketSender.
But if I change destination IP address from 255.255.255.255 to that of my PC, then it works.

rajeshdoshi, 2022-06-04:
Is there a way to receive broadcast messages by making any change in the library?

rajeshdoshi, 2022-06-04:
UDP broadcast

JAndrassy, 2022-06-04:
now I am confused. send and receive broadcast? receiving broadcast is turned off on ENC level by default with this library. I guess you changed the setting to receive broadcast and in your other issued you write it is limited to 18 bytes.

rajeshdoshi, 2022-06-05:
The observations have confused me as well. It seems UDP communication depends a lot on the traffic on the network. Today on a Sunday, when there are no other PCs connected on network, everything is working fine. 
As such observations are
1. I have not made in changes in the library. Still broadcast receipt works up to 18 characters maximum. 
1. ESP is able to receive maximum of 18 characters in Broadcast mode. 
2. Receipt of broadcast message seems to be dependent on traffic on the network. 
3. Broadcast messages sent from ESP seem to be working fine. 
4. When sending message from PC through packetSender directly to the IP address of ESP, if IP addresses of ESP and PC are not in same domain, ESP does not receive message. Even if I change the subnet to 0,0,0,0; ESP is not receiving the data.

martinez20m, 2023-01-26:
Just trying to enable udp broadcast receiving but with no luck. Anyone have any idea how to configure enc to receive udp broadcast packets?
My change (not working)@Enc28J60Network.cpp:
`  writeReg(ERXFCON, ERXFCON_UCEN|ERXFCON_CRCEN|ERXFCON_PMEN|ERXFCON_BCEN);
  writeRegPair(EPMM0, 0x303f);
  writeReg(EPMM2, 0x80);
  writeRegPair(EPMCSL, 0xe6ff);`

closed 2025-05-24 by JAndrassy