Networking-for-Arduino/EthernetENC #40: not pingable after a while

honk79, 2022-06-27:
Hi,
I use the library with the Nano Ethernet Shield on an ESP32 Dev Kit-C.
I also use the PubSubClient-Library to send MQTT messages.
After some time (varies between some seconds and half an hour) the shield won't answer on pings. The "LinkStatus" is still 1.
For me it looks like some buffer or memory problems. When I have less traffic, the problem takes longer to occure.

Thanks for the help!

JAndrassy, 2022-06-27:
try to add `Ethernet.maintain();` into loop()

honk79, 2022-06-27:
I tried it and placed it a the firts command in the loop(), but still the same problem.

JAndrassy, 2022-06-27:
does it execute at least every 200 milliseconds?

honk79, 2022-06-28:
yes, actually it is a very short loop.
`void loop() {
  Ethernet.maintain();
  if (!client.connected()) {
    mqttReconnect();
  }
  if (millis() - timeSensorLastUpdate >= timeSensorInterval) {
    timeSensorLastUpdate += timeSensorInterval;
    pir.readStatus();
    Serial.print("HardwareStatus: ");
    Serial.println(Ethernet.hardwareStatus());
    Serial.print("LinkStatus: ");
    Serial.println(Ethernet.linkStatus());
  }
  if (millis() - timeMQTTLastUpdate >= timeMQTTInterval) {
    timeMQTTLastUpdate += timeMQTTInterval;
    mqttSendAll();
  }
  client.loop();
}
`

JAndrassy, 2022-06-28:
and the sketch still runs when the enc28j60 doesn't respond to ping?

JAndrassy, 2022-06-28:
then I guess it is the problem that many small IoT devices (even commercial) have with home routers. the router forgets about them after a while.

bennydiamond, 2022-09-07:
I think I have the same issue. After a few days, my Uno with ENC28J60 becomes unresponsive but the sketch still runs.

Arduino UNO R3 with some generic ENC28J60 board.
Sketch runs the following (all lastest versions):
EthernetENC
PubSubClient
DSCKeybusInterface

Unfortunately, with all this code packed in, I cannot crank up the serial prints of EthernetENC to a useful level of verbosity.

JAndrassy, 2022-09-07:
could you try to reset the router. it may help?

bennydiamond, 2022-09-08:
It does not. Only resetting the Arduino resolves the issue, temporarily.

mikeycoding, 2023-07-04:
Same problem here. Using EthernetServer on Controllino Mini (identical to Arduino UNO) and the server is not reachable after a certain amount of runtime (after few hours). The rest of the program is running. Only resetting the arduino helps.

alisyd, 2024-01-22:
I am also facing the same issue, with Arduino 2560 and ENC28j60,  Connection goes deaf after a few hours of being active, Arduino works as a client to connect to a remote website and exchange messages every five seconds, so after a few hours of being active it loses connection and is not visible in the list of connected clients in router. I haven't tried building the custom delay function, which I will do shortly and test. I appreciate any help I can get as I am trying to use ENC28j60 in production and I have rest of the parts of this project working, idk if it is a good idea to use it in production, but if not this, I would like to know if there is anything else I should be using, thanks for any help in advance.

AakankshaS-ITSPE, 2025-06-10:
Hello, I am facing the same issue. there is nothing wrong with my code as it was working very well before. I ordered new hardware and ping doesn't work