Networking-for-Arduino/EthernetENC #46: Use pico-w with EthernetENC and WiFi problem

kestrel0907, 2022-11-24:
Hello,
I find the two problem:
1.If I use bool before or after the Ethernet.begin(mac) ,will make Ethernet.begin(mac) stuck.but if bool's  name first word is lowercase w~z Ethernet.begin(mac) will working.

like this ,Ethernet.begin(mac)  will stuck:
`EthernetState = true;
    if (Ethernet.begin(mac) == 0) {
    EthernetState = true;
    Serial.println("Failed to configure Ethernet using DHCP");
    // Check for Ethernet hardware present
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
      while (true) {
        delay(1); // do nothing, no point running without Ethernet hardware
      }
    }`

I went use pico-w WIFI AND enc28j60 ,my code is use Ethernet.linkStatus() to identify whether there is an Ethernet connection,if cable no connection use WIFI,But if put in the WiFi.begin(ssid, password) or WiFi.run() in setup or loop even without using,Ethernet.begin(mac) will stuck.

`#include <SPI.h>
#include <EthernetENC.h>
#include <WiFi.h>

#ifndef STASSID
#define STASSID "SSID"
#define STAPSK "STAPSK"
#endif
const char* ssid = STASSID;
const char* password = STAPSK;

const char* host = "djxmmx.net";
const uint16_t port = 17;
int status = WL_IDLE_STATUS; 

WiFiMulti multi;

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

char server[] = "www.google.com";    // name address for Google (using DNS)

// Set the static IP address to use if the DHCP fails to assign
IPAddress ip(192, 168, 6, 201);
IPAddress myDns(8, 8, 8, 8);

EthernetClient client;

// Variables to measure the speed
unsigned long beginMicros, endMicros;
unsigned long byteCount = 0;
bool printWebData = true;  // set to false for better speed measurement
bool EthernetState = false;
void setup() {
  // You can use Ethernet.init(pin) to configure the CS pin
  Ethernet.init(21);  // Most Arduino shields
 delay(3000);
  // Open serial communications and wait for port to open:
   
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // start the Ethernet connection:
  
   if (Ethernet.linkStatus() == LinkON)
  {
      Serial.println("Initialize Ethernet with DHCP:");
      if (Ethernet.begin(mac) == 0) {
      Serial.println("Failed to configure Ethernet using DHCP");
      // Check for Ethernet hardware present
      if (Ethernet.hardwareStatus() == EthernetNoHardware) {
        Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
        while (true) {
          delay(1); // do nothing, no point running without Ethernet hardware
        }
      }
      Serial.println("2");
      if (Ethernet.linkStatus() == LinkOFF) {
        Serial.println("Ethernet cable is not connected.");
      }
      // try to congifure using IP address instead of DHCP:
      Ethernet.begin(mac, ip, myDns);
      } else {
      Serial.print("  DHCP assigned IP ");
      Serial.println(Ethernet.localIP());
      }
  }
  else if (Ethernet.linkStatus() == LinkOFF)
  {
      while ( status != WL_CONNECTED) {
      Serial.print("Attempting to connect to WEP network, SSID: ");
      Serial.println(ssid);
      status = WiFi.begin(ssid, password);
      }
  }   
}

void loop() {
  
}`
I searched the library and found that the program would get stuck at DHCP

Thanks for any help

gurusivaguru, 2023-02-24:
hai 

 it is Ethernet.init(17);  // Most Arduino shield

gurusivaguru, 2023-02-24:
> Hello, I have the two problem:
> 
> 1. If I use bool before or after the Ethernet.begin(mac) ,will make Ethernet.begin(mac) stuck.but if bool's  name first word is lowercase w~z Ethernet.begin(mac) will working.
> 
> like this ,Ethernet.begin(mac) will stuck:
> 
> ```
> EthernetState = true;
> if (Ethernet.begin(mac) == 0) {
> EthernetState = true;
> Serial.println("Failed to configure Ethernet using DHCP");
> // Check for Ethernet hardware present
> if (Ethernet.hardwareStatus() == EthernetNoHardware) {
>   Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
>   while (true) {
>     delay(1); // do nothing, no point running without Ethernet hardware
>   }
> }
> ```
> 
> 2.I went use pico-w WIFI AND enc28j60 ,my code is use Ethernet.linkStatus() to identify whether there is an Ethernet connection,if cable no connection use WIFI,But if put in the WiFi.begin(ssid, password) or WiFi.run() in setup or loop even without using,Ethernet.begin(mac) will stuck.
> 
> ```
>     #include <SPI.h>
>     #include <EthernetENC.h>
>     #include <WiFi.h>
>     
>     #ifndef STASSID
>     #define STASSID "SSID"
>     #define STAPSK "STAPSK"
>     #endif
>     const char* ssid = STASSID;
>     const char* password = STAPSK;
>     
>     const char* host = "djxmmx.net";
>     const uint16_t port = 17;
>     int status = WL_IDLE_STATUS; 
>     
>     WiFiMulti multi;
>     
>     byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
>     
>     char server[] = "www.google.com";    // name address for Google (using DNS)
>     
>     // Set the static IP address to use if the DHCP fails to assign
>     IPAddress ip(192, 168, 6, 201);
>     IPAddress myDns(8, 8, 8, 8);
>     
>     EthernetClient client;
>     
>     // Variables to measure the speed
>     unsigned long beginMicros, endMicros;
>     unsigned long byteCount = 0;
>     bool printWebData = true;  // set to false for better speed measurement
>     bool EthernetState = false;
>     void setup() {
>       // You can use Ethernet.init(pin) to configure the CS pin
>       Ethernet.init(21);  // Most Arduino shields
>      delay(3000);
>       // Open serial communications and wait for port to open:
>        
>       Serial.begin(9600);
>       while (!Serial) {
>         ; // wait for serial port to connect. Needed for native USB port only
>       }
>     
>       // start the Ethernet connection:
>       
>        if (Ethernet.linkStatus() == LinkON)
>       {
>           Serial.println("Initialize Ethernet with DHCP:");
>           if (Ethernet.begin(mac) == 0) {
>           Serial.println("Failed to configure Ethernet using DHCP");
>           // Check for Ethernet hardware present
>           if (Ethernet.hardwareStatus() == EthernetNoHardware) {
>             Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
>             while (true) {
>               delay(1); // do nothing, no point running without Ethernet hardware
>             }
>           }
>           Serial.println("2");
>           if (Ethernet.linkStatus() == LinkOFF) {
>             Serial.println("Ethernet cable is not connected.");
>           }
>           // try to congifure using IP address instead of DHCP:
>           Ethernet.begin(mac, ip, myDns);
>           } else {
>           Serial.print("  DHCP assigned IP ");
>           Serial.println(Ethernet.localIP());
>           }
>       }
>       else if (Ethernet.linkStatus() == LinkOFF)
>       {
>           while ( status != WL_CONNECTED) {
>           Serial.print("Attempting to connect to WEP network, SSID: ");
>           Serial.println(ssid);
>           status = WiFi.begin(ssid, password);
>           }
>       }   
>     }
>     
>     void loop() {
>       
>     }
> ```
> 
> I searched the library and found that the program would get stuck at DHCP
> 
> Thanks for any help

hai

it is Ethernet.init(17); // Most Arduino shield