Networking-for-Arduino/EthernetENC #49: Request setting custom hostName

DRSDavidSoft, 2023-05-08:
Hi there, currently the hostname is hardcoded to ENC28J + the last three hex digits of the MAC address:

https://github.com/JAndrassy/EthernetENC/blame/master/src/Dhcp.cpp#L195-L201

I was wondering if it could be possible to set it to a custom value? Perhaps something akin to this PR that is for WIZnet's `Ethernet.h`?
https://github.com/arduino-libraries/Ethernet/pull/77

Great work BTW, I'm very impressed by how smooth this chip works. I didn't expect it to perform this good, so I appreciate your great library!

JAndrassy, 2023-05-08:
do you want to transfer the change to EthernetENC (it has the same DhcpClass) and submit a PR?

d2weber, 2024-09-08:
Didn't look at the code. But setting the hostname like this: `Ethernet.setHostname("myhostname");` before `Ethernet.begin(mac)` seems to work.

JAndrassy, 2024-09-08:
I added setHostname  Oct 3, 2023

closed 2024-09-08 by JAndrassy