Networking-for-Arduino/EthernetENC #51: trying to send broadcast SSDP message

fatmehosseini, 2023-06-08:
Hi, I am trying to send broadcast SSDP, based on this example
https://github.com/arduino-libraries/Ethernet/blob/master/examples/UDPSendReceiveString/UDPSendReceiveString.ino
by changing the IP to 239:255:255:250 and the port to 1900. 
But I don't receive any message on Wire Shark.
This is my code:
#include <EthernetENC.h>
#include <EthernetUdp.h>

#define UDP_TX_PACKET_MAX_SIZE 24

byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
IPAddress ip(239, 255, 255, 250);

unsigned int localPort = 1900;      // local port to listen on
char packetBuffer[UDP_TX_PACKET_MAX_SIZE];  // buffer to hold incoming packet,
char ReplyBuffer[] = "acknowledged";        // a string to send back
EthernetUDP Udp;

void setup() {
    Serial.begin(115200);
    Serial.println("START");
  Ethernet.init(25);  // Most Arduino shields
  Ethernet.begin(mac);
  if (Ethernet.hardwareStatus() == EthernetNoHardware) {
    Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
    while (true) {
      delay(1); // do nothing, no point running without Ethernet hardware
    }
  }
  if (Ethernet.linkStatus() == LinkOFF) {
    Serial.println("Ethernet cable is not connected.");
  }

  // start UDP
  Udp.begin(localPort);
  Serial.printf("local IP is %s \n", Ethernet.localIP().toString());
}

void loop() {
  // if there's data available, read a packet
  int packetSize = Udp.parsePacket();
  if (packetSize) 
  {
    Serial.print("Received packet of size ");
    Serial.println(packetSize);
    Serial.print("From ");
    IPAddress remote = Udp.remoteIP();
    for (int i=0; i < 4; i++) {
      Serial.print(remote[i], DEC);
      if (i < 3) {
        Serial.print(".");
      }
    }
    Serial.print(", port ");
    Serial.println(Udp.remotePort());

    // read the packet into packetBuffer
    Udp.read(packetBuffer, UDP_TX_PACKET_MAX_SIZE);
    Serial.println("Contents:");
    Serial.println(packetBuffer);

    // send a reply to the IP address and port that sent us the packet we received
  }
    Udp.beginPacket(ip, localPort);
    Udp.write(ReplyBuffer);
    Udp.endPacket();
  delay(10);
}

JAndrassy, 2023-06-08:
the example primarily shows receiving of UDP packets

fatmehosseini, 2023-06-08:
Udp.beginPacket(ip, localPort);
    Udp.write(ReplyBuffer);
    Udp.endPacket();
These three lines are out of if condition and UDP message is sending regardless if any messages are received or not, But it does not work.

fatmehosseini, 2023-06-10:
My problem solved, I was trying to send the message on 239, 255, 255, 250 IP which is a multicast IP and in your library multicast sending is disabled. I changed the IP to broadcast IP and my message are received on Wire-shark now. 
By the way, is there any way that I could enable multi-casting in this library?

JAndrassy, 2023-06-10:
the uIP TCP/IP stack library used as core for this library doesn't support multicast

closed 2025-05-24 by JAndrassy