Networking-for-Arduino/EthernetENC #53: Problem by traffic through switches

kobraevents, 2023-08-28:
the following structure:
ENC-----Managed Switch------ENC ->No Connection between devices
ENC-----Unmanged Switch------ENC ->Connection between devices

i use the libs with an ESP32 devkit

I have TP-Link and Netgear as Gigabit switches and two Fritzboxes. Everything runs stable for several days. Now the setup should change to its natural environment. This has a manageable Swtich for the purpose of Vlans. Nothing works anymore. I have several Linksys LGS310, HP, 3Com running here. PC, Internet, printer etc. runs over the switches without problems. I have also enabled all ports for UDP/TCP in the switches and set the ports to 10base/full or half. no change.
Tried ten times different ports without success. Also changed my code that actual sends and received udp traffic in server/client structure without any luck.

The helps in the internet do not bring me further:

- 100 to 120ohm between pin 1/2, 3/6.
- change to 10base full
- crossover cable

Also on the HP switch, Microsnes, Linksys or Cisco nothing works. I am at a loss. but the switches shows that there is traffic on the ports. Even on the old router, which can only handle 100base-t, the udp connection works.

JAndrassy, 2023-08-29:
try to revert the changes made in this commit
https://github.com/JAndrassy/EthernetENC/commit/9f3d44d7b038b2a80cb6d2a1b6396edab2d2a056

kobraevents, 2023-10-01:
please give me instructions, what should i do with these?
just download and replace in the lib folder?

JAndrassy, 2023-10-01:
edit your copy to the left/red version

cavbor, 2025-03-23:
There is a problem with the traffic between the networks. There are no problems in the UIPEthernet library.

 Laptop with Wireshark 
172.16.100.20/28 ----------------------------------------------------|
7c:8a:e1:14:82:ec                                                                             |
                                                                                                        |
Arduino Uno w NIC W5100                                                     Router Mikrotik                                                                        Laptop
172.16.100.19/28 ------------------------  172.16.100.17/28 Bridge (eth2, eth3, eth4, eth5 ), 10.10.10.1/24 Eth1 -----    10.10.10.10/24
de:10:e1:9c:01:01                                                                          |
                                                                                                      |
Arduino Nano w NIC ENC28J60                                                    |
172.16.100.18/28 ---------------------------------------------------
de:10:e1:01:9c:9c

Arduino with the EthernetENC library does not respond to PING and other requests between networks.
The laptop 172.16.100.20/28 freely pings Arduino Uno w NIC W5100 and Arduino Nano w NIC ENC28J60.
Laptop 10.10.10.10/24 can only ping Arduino Uno, Arduino Nano does not respond to ping.

[image]