Networking-for-Arduino/EthernetENC #55: no function `ethernet.MACAddress()` to drop-in replace ethernet.h

cyberboy666, 2023-10-05:
i am trying to use this library as a drop-in replacement for the standard [arduino ethernet library](https://github.com/arduino-libraries/Ethernet) inside the this [artnet library](https://github.com/hideakitai/ArtNet)

everything works perfectly - except for [this](https://github.com/hideakitai/ArtNet/blob/main/Artnet/ArtnetCommon.h#L624) line in artnet code where it calls:

```
#ifndef ESP8266
            Ethernet.MACAddress(mac);
#endif
```

which doesnt exist in the EthernetENC version. 

ultimately i want to add support for EthernetENC to the artnet library - although its not clear to me if i should solve this by modifying the logic in artnet or adding this function to this library ? is this meant to be used as drop-in like this ? is it possible to add this function here or a reason why its not here ? 

thanks for your help!

JAndrassy, 2023-10-06:
done

cyberboy666, 2023-10-06:
tested for my use case and it works! thanks again

closed 2023-10-06 by cyberboy666