Networking-for-Arduino/EthernetENC #56: Port Reuse on TCP Connections

N0W0RK, 2023-11-11:
In my development Process with frequent restarting, I experienced issues with working code breaking after resetting or flashing. 
After some Packet inspection with Wireshark, I figured out, that upon reset, the same sourcePort is used. THis however can be misinterpretedby the Server as an out of Sequence packet or simmilar.
This could also be the source of the issues described in #30 .

A simple Fix would be to set the `lastport` Variable randomly in `uip.c`.

JAndrassy, 2023-11-12:
I think it is only a problem of WireShark

N0W0RK, 2023-11-12:
My uvicorn server did not like this aswell. According to [RFC 9293](https://datatracker.ietf.org/doc/html/rfc9293#name-knowing-when-to-keep-quiet) Servers are supposed to keep quiet upon port and sequence number duplication. 

So it is not necessarily braking the protocoll but making life siginificantly harder.