Networking-for-Arduino/EthernetENC #60: Teensy 4.1 connection issues

ManBearPigWolf, 2024-02-08:
My work has a LED lighting controller that was designed around a Teensy 3.6 which receives a color command over Ethernet and controls APA102 LED strips. We recently built another batch of the controllers, but had to use Teensy 4.1's because 3.6's have been obsolete for quite a while now. The firmware originally used the UIPEthernet library, which we've removed and replaced with EthernetENC; but we can't get the Teensy 4.1 to connect. When we upload the firmware using EthernetENC to a 3.6 based controller, it connects and functions just fine. I can't provide our firmware because it's proprietary, but it really is about as basic as you can imagine. We have a ENC28J60-H by Olimex connected to pins 10-13, and we bit-bang the data/clock out to the APA102 on pins 2 & 3.

Our current theory is that the SPI connection isn't being set up correctly, so this may need to be an issue posted to Teensy's SPI library; since it appears the bus is initialized in the Enc28J60Network .cpp file. However, we aren't sure exactly where the problem lies so we just looking for any leads at the moment. Has there been any issues with Teensy 4.x boards before or is there a better library to use?

ManBearPigWolf, 2024-02-09:
I tried a bunch of different frequencies down to 1Mhz. At 1Mhz I can occasionally get it to ping, but it's not repeatable; more luck than anything. If it does start responding to pings, once I send it a command it seems to crash and needs to be reset (which could be a problem unrelated to this library...I don't even know if my core problem is with this library)

Anyway, because I can intermittently connect with a low SPI frequency I know the connections and hardware are solid. Should I log an issue with Teensy's SPI library at this point? Or do you have any other ideas?

JAndrassy, 2024-02-09:
I have no other idea. The library works on many platforms so why wouldn't it work on Teensy 4.1. 
Doesn't have Teensy 4.1 an in-MCU ETH peripheral? The QNEthernet library is very good. [I tested it](https://github.com/ssilverman/QNEthernet/discussions/62) ported to STM32 with a W5500. (and it can be used the same way with an enc28j60))

ManBearPigWolf, 2024-02-12:
The 4.1 has a default clock speed of 528MHz in the IDE, while the 3.6 board has a default speed of 180MHz. I changed the CPU speed on the 4.1 to 150MHz and I can ping/connect to it just fine. I don't know why this works but for the moment I'm just happy. Anyway, if you're going from T3.6 to T4.1 check your CPU speeds!