Networking-for-Arduino/EthernetENC #69: connecting the Ethernet Client to the Server on Same ENC28J60 Device Fails

fatmehosseini, 2024-12-16:
I am encountering an issue when trying to run both an Ethernet server and an Ethernet client on the same ESP32 device using the EthernetENC library with an ENC28J60 module. While individually, both the client and server functionality work as expected, they fail to operate simultaneously. 

Observed Behavior:
- The client fails to connect to the server running on the same ENC28J60-based ESP32.
- The server does not register the incoming connection.
- Debugging suggests packets are not properly routed back to the server.

Expected Behavior:
The Ethernet client and server on the same ESP32 should be able to communicate over the ENC28J60 hardware.

- Hardware: ESP32 with ENC28J60 Ethernet Module
- Network Setup: Direct connection to ENC28J60 without an external switch/router (client and server on same device).

I would appreciate your thoughts on addressing this limitation or guidance on implementing a workaround within the library.

JAndrassy, 2024-12-16:
are you using the same EthernetClient for both?

fatmehosseini, 2024-12-16:
No I use different ones
the EhternetClient that I am using to respond to server is a local variable in a function and the one that I send request is also local in another function 
but the problem is Client.connect(Ethernet.localIP(), port) returns 0

JAndrassy, 2024-12-16:
uIP can't connect to itself

fatmehosseini, 2024-12-16:
connecting to itself is able when the device uses wifi, with HTTPClient library 
is there any way that I can do this on ethernet too?
the reason that I am doing it is that I have a http server which goes down after a while I am trying to check if the server still works and in case of going down reset the network connection 
when the server goes down pinging the IP does not work either but a UDP message that the device broadcasts periodically still works

JAndrassy, 2024-12-16:
you can use my EthernetESP32 library with ESP32 Arduino version 3

closed 2025-05-24 by JAndrassy