Networking-for-Arduino/EthernetENC #72: Library doesn't read UDP packages!

ruiseixasm, 2025-05-19:
I'm trying to make this sketch work, the sent part of the loop works fine, but the receive part never picks anything! The ping on the device works just fine, so, it's not a problem of the network.

```
#include <EthernetENC.h>
#include <EthernetUdp.h>

// // Linux testing commands:
// echo "BROADCAST 255" | nc -ubv 255.255.255.255 5005
// echo "BROADCAST 192" | nc -ubv 192.168.31.255 5005
// echo "UNICAST" | nc -ubv 192.168.31.100 5005

// Network Settings
byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
IPAddress ip(192, 168, 31, 100);       // Arduino IP
IPAddress subnet(255, 255, 255, 0);   // Network mask
IPAddress gateway(192, 168, 31, 77);    // Router IP (if needed)
const unsigned int UDP_PORT = 5005;
IPAddress target(192, 168, 31, 22);                // Target PC IP
IPAddress broadcast(255, 255, 255, 255); // Broadcast address

EthernetUDP udp;
unsigned long lastSendTime = 0;
const unsigned long INTERVAL = 5000; // 5 seconds

void setup() {
    Serial.begin(9600);
    
    // Initialize Ethernet with static IP
    Ethernet.begin(mac, ip, gateway, subnet);

    // Start UDP
    if (udp.begin(UDP_PORT)) {
        Serial.print("\n\nUDP active on ");
        Serial.println(Ethernet.localIP());
    } else {
        Serial.println("UDP failed!");
    }
}

void loop() {

    // Send "Hello" every 5 seconds
    if (millis() - lastSendTime >= INTERVAL) {
        if (random(2) % 2 == 0) {
            Serial.println("Direct hello");
            udp.beginPacket(target, UDP_PORT);
            udp.write("Direct hello");
        } else {
            Serial.println("Broadcasted hello");
            udp.beginPacket(broadcast, UDP_PORT);
            udp.write("Broadcasted hello");
        }
        udp.endPacket();
        lastSendTime = millis();
    }

    
    // Check for incoming packets
    int packetSize = udp.parsePacket();
    if (packetSize > 0) {
        char packet[128];
        udp.read(packet, sizeof(packet));
        Serial.print("From ");
        Serial.print(udp.remoteIP());
        Serial.print(": ");
        Serial.println(packet);
    }
}
```

However, the received packages in the other machine are these:

```
b'Direct helloBroadca'
b'Broadcasted hello'
b'Direct helloBroadca'
b'Direct helloBroadca'
b'Broadcasted hello'
b'Broadcasted hello'
b'Direct helloDirect Direct Direct Direct Direct Direct Broadca'
b'Direct helloDirect Broadca'
b'Direct helloBroadca'
b'Broadcasted hello'
b'Broadcasted hello'
b'Direct helloDirect Broadca'
b'Direct helloBroadca'
b'Broadcasted hello'
```

And the Arduino nano never receives any package I send to it, regardless from the Python script or from these commands:

```
echo "BROADCAST 255" | nc -ubv 255.255.255.255 5005
echo "BROADCAST 192" | nc -ubv 192.168.31.255 5005
echo "UNICAST" | nc -ubv 192.168.31.100 5005
```

Ping works just fine, so it must be a problem with UDP implementation!

```
Pinging 192.168.31.100 with 32 bytes of data:
Reply from 192.168.31.100: bytes=32 time=1ms TTL=128
Reply from 192.168.31.100: bytes=32 time=1ms TTL=128
Reply from 192.168.31.100: bytes=32 time=1ms TTL=128
Reply from 192.168.31.100: bytes=32 time=1ms TTL=128

Ping statistics for 192.168.31.100:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 1ms, Maximum = 1ms, Average = 1ms
```

Can you help me in make the script work, these is the compiling output.

```
Sketch uses 16456 bytes (53%) of program storage space. Maximum is 30720 bytes.
Global variables use 1054 bytes (51%) of dynamic memory, leaving 994 bytes for local variables. Maximum is 2048 bytes.
```
Thanks

JAndrassy, 2025-05-19:
https://github.com/Networking-for-Arduino/EthernetENC/wiki/Limitations

ruiseixasm, 2025-05-19:
By limitations does that means that contrary to the [EtherCard](https://github.com/njh/EtherCard) the EthernetENC doesn't capture UDP packages?

JAndrassy, 2025-05-20:
> To reduce the load, ENC28J60 can filter out packets with very simple rules and don't store them for processing. This way the library filters out broadcast packets and allows only packets with the given MAC address. 

broadcast UDP

ruiseixasm, 2025-05-21:
So, that means I should use the [UIPEthernet](https://docs.arduino.cc/libraries/uipethernet/) library instead, like this:
```
#include <UIPEthernet.h>
#include <UIPUdp.h>  // If using UDP
```

JAndrassy, 2025-05-21:
what is the source of the UDP broadcast? does it really have to be a broadcast?

ruiseixasm, 2025-05-23:
Yes, it has, the Devices start to interact by broadcasting messages and only then they switch to Unicast once responding to them. It uses JSON as encoded communication with checksum included.

CoyoteProd, 2025-05-23:
I'm in the same need. 
I have to read WoL Broadcast on port 9

JAndrassy, 2025-05-23:
```
  writeReg(ERXFCON, ERXFCON_UCEN|ERXFCON_CRCEN|ERXFCON_PMEN|ERXFCON_BCEN);
```

ruiseixasm, 2025-05-24:
Ok, I found it in the datasheet:
```
UCEN: Unicast Filter Enable bit
CRCEN: Post-Filter CRC Check Enable bit
PMEN: Pattern Match Filter Enable bit
MPEN: Magic Packet Filter Enable bit
HTEN: Hash Table Filter Enable bit
MCEN: Multicast Filter Enable bit
BCEN: Broadcast Filter Enable bit
```